rpng/MINS #1: Lidar Undistortion

lnexenl, 2023-10-09:
Hi, I currently run MINS on some private datasets, where a car goes around the figure eight, requiring a high-precision lidar undistortion function. 

However, I find that [lidar undistortion function](https://github.com/rpng/MINS/blob/9642045a9fb7a2a239799f749820c653e8e454ff/mins/src/update/lidar/LidarHelper.cpp#L39) requires v_angles and h_angles to undistort point cloud. And this function also assumes that the time differences between each point are same.
`success = state->get_interpolated_pose(lidar_inL->time + dt - (total - i) * op->raw_point_dt, RGtoIi, pIiinG);`

I think lidar undistortion function could be made more general, by getting each point's timestamp directly from ros msg. Is it possible to introduce this as a new feature in your future work?

WoosikLee2510, 2023-10-09:
Hi, thanks for having interest in this project and providing your thoughts.

The code is written pretty easily to be modified for your application.
You need to pass the point's timestamp you get from the message to this function and replace the [lidar time](https://github.com/rpng/MINS/blob/9642045a9fb7a2a239799f749820c653e8e454ff/mins/src/update/lidar/LidarHelper.cpp#L75C73-L75C73) input with it (add 'dt' if lidar timeoffset is non-zero).

I assume you are using Livox kind of LiDAR sensors which provide each point's timestamp. I didn't implement the function in the way you suggested because most other widely used sensors, such as Velodyne or Ouster LiDARs do not provide such detailed information. I will put this on my todo list.

lnexenl, 2023-10-09:
Yes, the dataset includes some solid-state lidars. May be I will do some work on this submit a pull request later.

WoosikLee2510, 2023-10-09:
Sure, your contribution will be welcomed. Please reopen the issue if you need any other thing to discuss.

closed 2023-10-09 by WoosikLee2510

mahammadirfan, 2024-03-14:
@lnexenl hey have you tried MINS with any livox lidar ?

lnexenl, 2024-03-15:
> @lnexenl hey have you tried MINS with any livox lidar ?

I haven't tried livox lidar, I worked on robosense and seyond lidars. These lidars produce pointclouds with timestamp on each point. But the undistortion method currently used in MINS doesn't suit these lidars very well. And I finally use MINS as a multi-cam/INS/Wheel odometry.

goldtimes, 2024-09-20:
@lnexenl have you updated the lidar undistortion function?

lnexenl, 2024-09-20:
> @lnexenl have you updated the lidar undistortion function?

No, I didn't work on MINS's Lidar SLAM anymore, it's not easy to modify this.