rpng/MINS #5: [KAIST] Process died when use_imu_res is disabled.

SchwartzSkipper, 2023-10-11:
Hi, Woosik. Really appreciate this great work!

I am currently running KAIST urban 30 dataset. Everything works fine when using subscribe.launch with the defaut kaist/kaist_LC config.

However, when I try to run with polynomial estimated residuals,  
`mins/config/kaist/kaist_LC/config_estimator.yaml`-> `use_imu_res: false`,
the `[mins_subscribe-2]` process died on startup. 
[image]

Could you take a look at this problem? Much appreciated.

WoosikLee2510, 2023-10-12:
Hi, thanks for using MINS!
I fixed the issue with [commit](00ea4229bfe7ebaa373f090acd935342506ea923).
Try pulling the latest version and let me know if it fixes your problem.

SchwartzSkipper, 2023-10-13:
Thanks! 
This problem is solved.  😃👍

closed 2023-10-13 by SchwartzSkipper