rpng/MINS #6: Random crash of program

lnexenl, 2023-10-12:
Hi, I find that MINS really works well, but sometimes it crashes my system, causing a reboot of my PC (no matter where I execute the program: inside or outside docker container). Have you ever met such problem? I am curious whether it's a common problem or it only happens on my PC. I guess this problem might be caused by invalid memory access.

WoosikLee2510, 2023-10-12:
I haven't seen a random crash caused by MINS so far. I used 2 PCs to run evaluations(in the paper) and 1 for build and other compatibility tests. 

What's the config you are running? What combinations of sensors are you using? Are you running simulation, rosbag, or subsruber?

lnexenl, 2023-10-12:
I am running rosbag with some private bags including 1 cams and 1 IMU. I also launch rviz in the launch file. Maybe rviz is related to the crash? I will conduct some tests on another PC later.

lnexenl, 2023-10-18:
I run MINS 20 times continuously on my another PC, it doesn't crash. I believe there is definitely some problem with my PC.

WoosikLee2510, 2023-10-18:
Thanks for the check.