rpng/MINS #7: Is there a ROS2 version?

mzahana, 2023-10-14:
Thanks for the great work.

Is there a plan to provide ROS2 support ?

WoosikLee2510, 2023-10-14:
Hi, supporting ROS2 is on todo list. Not sure exactly when I can add the feature, but hopefully soon. I expect the implementation should be similar to how OpenVINS is supporting it.

mzahana, 2023-10-15:
@WoosikLee2510 That would be great!
Thank you.

mzahana, 2024-02-09:
@WoosikLee2510 any updates?

WoosikLee2510, 2024-02-26:
@mzahana slowly working on it. Sorry for the delay. If you want to try it yourself, you can refer to [OpenVINS](https://github.com/rpng/open_vins) to grab how you can implement it - which is what I am working on.

sacovo, 2024-03-12:
Is there any way to assist you in porting it to ros2? I would like to use the project in our ros2 system, and also have some time to spend on porting it to ros2. So if you have a branch where you already started that would be great, otherwise I could give it a go.

Are there larger things that need to change in order to port it to ros2, or is it more or less just the things mentioned in [here](https://docs.ros.org/en/humble/How-To-Guides/Migrating-from-ROS1/Migrating-CPP-Packages.html)?

sacovo, 2024-03-16:
I have something that builds and also starts, the simulation also looks good. But testing it with euroc_mav datasets doesn't quite work. I don't understand the math behind the whole thing good enough to figure out, what's going wrong, so if you could have a look at it @WoosikLee2510, that would be great!

I started a pull request in #25

mzahana, 2024-04-23:
@WoosikLee2510 Any updates on this ?

WoosikLee2510, 2024-05-03:
Hi, @mzahana . I am truly sorry for sitting on this update so long. How is [#25](https://github.com/rpng/MINS/pull/25)? I will try to look into it too... And much appreciated to @sacovo !

sacovo, 2024-05-08:
I think with the QoS thing sorted out it works with ROS2, I get good results with the euroc_mav datasets. I think the code needs some cleaning up, and I think I'll get to that in the next few weeks. @WoosikLee2510 if you have the time to add some comments for me to review I'll have a look at them.

sacovo, 2024-08-07:
I created a cleaner port: #38

mzahana, 2024-09-17:
Thanks @sacovo 
Does your ros2 branch support GPS and LIDAR ?

sacovo, 2024-09-23:
We are not using those two sensors on our platform so I can't tell you for sure. In the simulation (with ros2 run mins simulation) it seemed to work with the simulated lidar input.

I think the kitty dataset has both of those, so it might be possible to test it with those datasets.

Regarding Lidar, do you want to use a 2d or a 3d lidar? As far as I can tell the package expects a point cloud, instead of a laser scan, so I think another conversion from ROS message to internal datatype would be required.

mzahana, 2024-09-23:
@sacovo Thanks for your reply. I would like to fuse 3D LiDAR+GPS+camera+IMU. I will test and share feedback.