rpng/MINS #8: Can MINS work on a drone?

mzahana, 2023-10-16:
Hi!

I read the paper of MINS, and I see the experiments were done with ground vehicles. I would assume it is usable on a drone, but I wanted to confirm this from a practical point of view. Would there be specific assumptions for MINS to be used one e.g., a multi-rotor UAV?

Thanks.

WoosikLee2510, 2023-10-16:
Hi, MINS can be used on most existing platforms including grounded vehicles and drones. For example, you can run visual-inertial,  visual-gps-inertial, visual-lidar-inertial, or lidar-gps-inertial using MINS on drone. 

The paper used a grounded vehicle dataset because MINS can also fuse wheel measurement.

WoosikLee2510, 2023-10-16:
If you are specifically looking into fusing multiple rotors with other sensors, e.g., VIO, then you can also look into our [prior work](https://woosiklee2510.github.io/downloads/papers/Chen2022IROS.pdf)

WoosikLee2510, 2023-10-18:
Please open this issue again or add a comment if you have any further questions about MINS.

closed 2023-10-18 by WoosikLee2510