rpng/MINS #10: Sensor usage consultation

WuZihao12, 2023-10-17:
Does this project support multi-camera mode like [MIMC-VINS](https://ieeexplore.ieee.org/abstract/document/9363450) ?

WoosikLee2510, 2023-10-17:
Yes, it does. 
You can add as many cameras as you want (for example, it can run ladybug camera) and also can do full online calibration of each camera including time offset, extrinsic, and intrinsic. The only computational increase due to having more cameras would be feature extraction and update (if you do more updates with a larger number of features).
MINS also can freely choose stereo camera pairs or multiple mono cameras. 
Check [camera options](https://github.com/rpng/MINS/blob/master/mins/config/simulation/config_camera.yaml) for more detailed availabilities.
Note, that MINS does not support multi-IMU, though can be easily added.

closed 2023-10-17 by WoosikLee2510

WuZihao12, 2023-10-17:
This project is equivalent to also using on-manifold state interpolation technique to achieve asynchronous between multiple cameras?

WuZihao12, 2023-10-17:
> Yes, it does. You can add as many cameras as you want (for example, it can run ladybug camera) and also can do full online calibration of each camera including time offset, extrinsic, and intrinsic. [...]

This project is equivalent to also using on-manifold state interpolation technique to achieve asynchronous between multiple cameras?

WoosikLee2510, 2023-10-17:
Yes, it is. It uses the same interpolation technique of MIMC-VINS to handle multiple cameras which is the reason why it can handle multiple cameras without increasing the state size (meaning without having clones to each camera measurement times) gaining computational efficiency.

WuZihao12, 2023-10-17:
Thanks for your reply, thank you !