rpng/MINS #9: Project Consulting

WuZihao12, 2023-10-17:
Thank you for your great work. What I want to ask is what is the relationship between this project and [MIMC-VINS](https://ieeexplore.ieee.org/document/9363450)？

WoosikLee2510, 2023-10-17:
MIMC-VINS first proposed a novel on-manifold state interpolation technique. 
MINS adapts the technique and also addresses the error caused by interpolation which improves the system consistency. 
Additionally, MINS proposed a dynamic cloning technique based on the investigation that allows large computational efficiency without losing much accuracy. You can check more details in the [paper section 4](https://woosiklee2510.github.io/downloads/papers/Lee2024IJRR.pdf)

closed 2023-10-17 by WoosikLee2510

WuZihao12, 2023-10-17:
Does this project support multi-camera mode like [MIMC-VINS](https://ieeexplore.ieee.org/abstract/document/9363450) ?