rpng/MINS #13: process[mins simulation0-1]: started with pid [7471]

ZJJ2022415, 2023-11-18:
Excuse me, I am running "roslaunch mins simulation.launch cam_enabled:=true lidar_enabled:=true". What is the cause of the error? How to fix it?

[image]

And I'm using "roslaunch mins serial.launch config:=kaist/kaist_LC path_gt:=urban30.txt path_bag:=urban30.bag "When running rosbag with your own kaist dataset, can the bag contain only the following topics? What are the topics of your bag?

[image]

Thank you very much.

WoosikLee2510, 2023-11-18:
Hi @ZJJ2022415 .
For the simulation, the process died without any messages. Can you try running it with `sys_verbosity=true` and share what you get?

In order to run MINS with kaist lc config, you need have rosbag that contains camera and lidar messages. It looks like your bag doesn't have it. I am not sure where you got the bag but I recommend downloading the sensor data from the original website: https://sites.google.com/view/complex-urban-dataset/download-lidar-stereo

You can convert the sensor data to bag using packages, e.g., https://github.com/tsyxyz/kaist2bag

ZJJ2022415, 2023-11-18:
Thank you. I'll recreate the bag. After running' roslaunch mins simulation.launch sys_verbosity:=true' I get the following result:

(base) zjj@zjj:~/zjj/MINS01_ws$ roslaunch mins simulation.launch sys_verbosity:=true
... logging to /home/zjj/.ros/log/8f50d230-85bb-11ee-9050-89c65a8969a2/roslaunch-zjj-8393.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
RLException: Invalid <node> tag: Invalid <param> tag: invalid literal for int() with base 10: 'true'. 
Param xml is <param name="sys_verbosity" type="int" value="$(arg sys_verbosity)"/>. 
Node xml is <node name="mins_simulation$(arg sim_seed)" pkg="mins" type="simulation" output="screen" clear_params="true" required="true">

        <param name="config_path" type="str" value="$(arg config_path)"/>
        <param name="sys_verbosity" type="int" value="$(arg sys_verbosity)"/>
        <param name="sys_exp_id" type="int" value="$(arg sys_exp_id)"/>
        <param name="sys_save_state" type="bool" value="$(arg sys_save_state)"/>
        <param name="sys_path_state" type="str" value="$(arg sys_path_state)"/>
        <param name="sys_save_timing" type="bool" value="$(arg sys_save_timing)"/>
        <param name="sys_path_timing" type="str" value="$(arg sys_path_timing)"/>
        <param name="sys_save_trajectory" type="bool" value="$(arg sys_save_trajectory)"/>
        <param name="sys_path_trajectory" type="str" value="$(arg sys_path_trajectory)"/>
        <param name="sim_BSpline_path" type="str" value="$(arg sim_BSpline_path)"/>
        <param name="sim_planes_path" type="str" value="$(arg sim_planes_path)"/>
        <param name="sim_seed" type="int" value="$(arg sim_seed)"/>
        <param name="sim_n_loop" type="int" value="$(arg sim_n_loop)"/>
        <param name="sim_const_planar" type="bool" value="$(arg sim_const_planar)"/>
        <param name="sim_const_holonomic" type="bool" value="$(arg sim_const_holonomic)"/>
        <param name="sim_do_perturb" type="bool" value="$(arg sim_do_perturb)"/>
        <param name="sim_remove_noise" type="bool" value="$(arg sim_remove_noise)"/>
        <param name="sim_freq_imu" type="double" value="$(arg sim_freq_imu)"/>
        <param name="sim_freq_cam" type="double" value="$(arg sim_freq_cam)"/>
        <param name="sim_freq_gps" type="double" value="$(arg sim_freq_gps)"/>
        <param name="sim_freq_wheel" type="double" value="$(arg sim_freq_wheel)"/>
        <param name="sim_freq_vicon" type="double" value="$(arg sim_freq_vicon)"/>
        <param name="sim_freq_lidar" type="double" value="$(arg sim_freq_lidar)"/>
        <param name="init_use_gt_gnss" type="bool" value="$(arg init_use_gt_gnss)"/>
        <param name="init_use_gt_lidar" type="bool" value="$(arg init_use_gt_lidar)"/>
        
        <param name="est_window_size" type="double" value="$(arg est_window_size)"/>
        <param name="est_clone_freq" type="int" value="$(arg est_clone_freq)"/>
        <param name="est_intr_order" type="int" value="$(arg est_intr_order)"/>
        <param name="est_dynamic_cloning" type="bool" value="$(arg est_dynamic_cloning)"/>
        <param name="est_use_imu_res" type="bool" value="$(arg est_use_imu_res)"/>
        <param name="est_use_imu_cov" type="bool" value="$(arg est_use_imu_cov)"/>
        <param name="est_use_pol_cov" type="bool" value="$(arg est_use_pol_cov)"/>
        <param name="est_intr_error_mlt" type="double" value="$(arg est_intr_error_mlt)"/>
        <param name="est_intr_error_ori_thr" type="double" value="$(arg est_intr_error_ori_thr)"/>
        <param name="est_intr_error_pos_thr" type="double" value="$(arg est_intr_error_pos_thr)"/>
        <param name="est_intr_error_thr_mlt" type="double" value="$(arg est_intr_error_thr_mlt)"/>
        
        <param name="cam_enabled" type="bool" value="$(arg cam_enabled)"/>
        <param name="cam_use_stereo" type="bool" value="$(arg cam_use_stereo)"/>
        <param name="cam_max_n" type="int" value="$(arg cam_max_n)"/>
        <param name="cam_max_slam" type="int" value="$(arg cam_max_slam)"/>
        <param name="cam_do_calib_ext" type="bool" value="$(arg cam_do_calib_ext)"/>
        <param name="cam_do_calib_int" type="bool" value="$(arg cam_do_calib_int)"/>
        <param name="cam_do_calib_dt" type="bool" value="$(arg cam_do_calib_dt)"/>
        <param name="cam_chi2_mult" type="double" value="$(arg cam_chi2_mult)"/>
        <param name="cam_init_cov_dt" type="double" value="$(arg cam_init_cov_dt)"/>
        <param name="cam_init_cov_ex_o" type="double" value="$(arg cam_init_cov_ex_o)"/>
        <param name="cam_init_cov_ex_p" type="double" value="$(arg cam_init_cov_ex_p)"/>
        <param name="cam_init_cov_in_k" type="double" value="$(arg cam_init_cov_in_k)"/>
        <param name="cam_init_cov_in_c" type="double" value="$(arg cam_init_cov_in_c)"/>
        <param name="cam_init_cov_in_r" type="double" value="$(arg cam_init_cov_in_r)"/>
        
        <param name="gps_enabled" type="bool" value="$(arg gps_enabled)"/>
        <param name="gps_max_n" type="int" value="$(arg gps_max_n)"/>
        <param name="gps_chi2_mult" type="double" value="$(arg gps_chi2_mult)"/>
        <param name="gps_do_calib_ext" type="bool" value="$(arg gps_do_calib_ext)"/>
        <param name="gps_do_calib_dt" type="bool" value="$(arg gps_do_calib_dt)"/>
        <param name="gps_init_cov_dt" type="double" value="$(arg gps_init_cov_dt)"/>
[... 11 lines cut ...]
[... cut ...]
        <param name="vicon_do_calib_dt" type="bool" value="$(arg vicon_do_calib_dt)"/>
        
        <param name="imu_accel_noise" type="double" value="$(arg imu_accel_noise)"/>
        <param name="imu_accel_bias" type="double" value="$(arg imu_accel_bias)"/>
        <param name="imu_gyro_noise" type="double" value="$(arg imu_gyro_noise)"/>
        <param name="imu_gyro_bias" type="double" value="$(arg imu_gyro_bias)"/>
    </node>
The traceback for the exception was written to the log file

WoosikLee2510, 2023-11-19:
I am sorry, 'sys_verbosity=0'

ZJJ2022415, 2023-11-19:
Thanks. I get the following result:

(base) zjj@zjj:~/zjj/MINS01_ws$ roslaunch mins simulation.launch sys_verbosity:=0
... logging to /home/zjj/.ros/log/2c73a7dc-86a1-11ee-a6d6-05eb00dc89b1/roslaunch-zjj-4242.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://zjj:33409/

SUMMARY
========

CLEAR PARAMETERS
 * /mins_simulation0/

PARAMETERS
 * /mins_simulation0/cam_chi2_mult: 1.0
 * /mins_simulation0/cam_do_calib_dt: False
 * /mins_simulation0/cam_do_calib_ext: False
 * /mins_simulation0/cam_do_calib_int: False
 * /mins_simulation0/cam_enabled: True
 * /mins_simulation0/cam_init_cov_dt: 0.0001
 * /mins_simulation0/cam_init_cov_ex_o: 0.0001
 * /mins_simulation0/cam_init_cov_ex_p: 0.001
 * /mins_simulation0/cam_init_cov_in_c: 1.0
 * /mins_simulation0/cam_init_cov_in_k: 1.0
 * /mins_simulation0/cam_init_cov_in_r: 1e-05
 * /mins_simulation0/cam_max_n: 2
 * /mins_simulation0/cam_max_slam: 25
 * /mins_simulation0/cam_use_stereo: True
 * /mins_simulation0/config_path: /home/zjj/zjj/MIN...
 * /mins_simulation0/est_clone_freq: 20
 * /mins_simulation0/est_dynamic_cloning: False
 * /mins_simulation0/est_intr_error_mlt: 1.0
 * /mins_simulation0/est_intr_error_ori_thr: 0.007
 * /mins_simulation0/est_intr_error_pos_thr: 0.003
 * /mins_simulation0/est_intr_error_thr_mlt: 1.0
 * /mins_simulation0/est_intr_order: 3
 * /mins_simulation0/est_use_imu_cov: False
 * /mins_simulation0/est_use_imu_res: True
 * /mins_simulation0/est_use_pol_cov: True
 * /mins_simulation0/est_window_size: 1.0
 * /mins_simulation0/gps_chi2_mult: 1.0
 * /mins_simulation0/gps_do_calib_dt: False
 * /mins_simulation0/gps_do_calib_ext: False
 * /mins_simulation0/gps_enabled: False
 * /mins_simulation0/gps_init_cov_dt: 0.001
 * /mins_simulation0/gps_init_cov_ex: 0.001
 * /mins_simulation0/gps_max_n: 2
 * /mins_simulation0/gps_noise: 0.1
 * /mins_simulation0/imu_accel_bias: 0.03
 * /mins_simulation0/imu_accel_noise: 0.02
 * /mins_simulation0/imu_gyro_bias: 0.00019393
 * /mins_simulation0/imu_gyro_noise: 0.0016968
 * /mins_simulation0/init_use_gt_gnss: True
 * /mins_simulation0/init_use_gt_lidar: True
 * /mins_simulation0/lidar_chi2_mult: 1.0
 * /mins_simulation0/lidar_do_calib_dt: False
 * /mins_simulation0/lidar_do_calib_ext: False
 * /mins_simulation0/lidar_enabled: True
 * /mins_simulation0/lidar_init_cov_dt: 0.0001
 * /mins_simulation0/lidar_init_cov_ex_o: 0.0001
 * /mins_simulation0/lidar_init_cov_ex_p: 0.001
 * /mins_simulation0/lidar_map_downsample_size: 0.3
 * /mins_simulation0/lidar_max_n: 1
 * /mins_simulation0/lidar_raw_downsample_size: 0.3
 * /mins_simulation0/sim_BSpline_path: /home/zjj/zjj/MIN...
 * /mins_simulation0/sim_const_holonomic: True
 * /mins_simulation0/sim_const_planar: False
 * /mins_simulation0/sim_do_perturb: True
 * /mins_simulation0/sim_freq_cam: 30.0
 * /mins_simulation0/sim_freq_gps: 1.0
 * /mins_simulation0/sim_freq_imu: 200.0
 * /mins_simulation0/sim_freq_lidar: 10.0
 * /mins_simulation0/sim_freq_vicon: 20.0
 * /mins_simulation0/sim_freq_wheel: 100.0
 * /mins_simulation0/sim_n_loop: 0
 * /mins_simulation0/sim_planes_path: /home/zjj/zjj/MIN...
 * /mins_simulation0/sim_remove_noise: False
 * /mins_simulation0/sim_seed: 0
 * /mins_simulation0/sys_exp_id: 0
 * /mins_simulation0/sys_path_state: /home/zjj/zjj/MIN...
 * /mins_simulation0/sys_path_timing: /home/zjj/zjj/MIN...
 * /mins_simulation0/sys_path_trajectory: /home/zjj/zjj/MIN...
 * /mins_simulation0/sys_save_state: True
 * /mins_simulation0/sys_save_timing: True
 * /mins_simulation0/sys_save_trajectory: True
 * /mins_simulation0/sys_verbosity: 0
 * /mins_simulation0/vicon_do_calib_dt: False
 * /mins_simulation0/vicon_do_calib_ext: False
 * /mins_simulation0/vicon_enabled: False
 * /mins_simulation0/vicon_max_n: 2
 * /mins_simulation0/wheel_chi2_mult: 1.0
 * /mins_simulation0/wheel_do_calib_dt: False
 * /mins_simulation0/wheel_do_calib_ext: False
 * /mins_simulation0/wheel_do_calib_int: False
 * /mins_simulation0/wheel_enabled: False
 * /mins_simulation0/wheel_init_cov_dt: 0.0001
 * /mins_simulation0/wheel_init_cov_ex_o: 0.0001
 * /mins_simulation0/wheel_init_cov_ex_p: 0.001
 * /mins_simulation0/wheel_init_cov_in_b: 0.0001
 * /mins_simulation0/wheel_init_cov_in_r: 0.0001
 * /rosdistro: noetic
 * /rosversion: 1.15.15

NODES
  /
    mins_simulation0 (mins/simulation)

auto-starting new master
process[master]: started with pid [4263]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 2c73a7dc-86a1-11ee-a6d6-05eb00dc89b1
process[rosout-1]: started with pid [4284]
started core service [/rosout]
process[mins_simulation0-2]: started with pid [4288]
================================================================================REQUIRED process [mins_simulation0-2] has died!
process has died [pid 4288, exit code -11, cmd /home/zjj/zjj/MINS01_ws/devel_isolated/mins/lib/mins/simulation __name:=mins_simulation0 __log:=/home/zjj/.ros/log/2c73a7dc-86a1-11ee-a6d6-05eb00dc89b1/mins_simulation0-2.log].
log file: /home/zjj/.ros/log/2c73a7dc-86a1-11ee-a6d6-05eb00dc89b1/mins_simulation0-2*.log
Initiating shutdown!
================================================================================
[mins_simulation0-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

WoosikLee2510, 2023-11-19:
Hmm.. there is nothing that comes out from the log - except that the mins/simulation node is initialized and died. Can you try checking where it actually dies? It usually should print out something.

ZJJ2022415, 2023-11-21:
I changed my computer and it worked successfully. What are the rostopics of your kaist dataset, because the dataset I merged seems to be a bit problematic.

closed 2023-11-21 by ZJJ2022415

WoosikLee2510, 2023-11-21:
The topics should be what I used in the config files:
--
/stereo/left/image_raw
/stereo/right/image_raw
/gps/fix
/imu/data_raw
/ns1/velodyne_points
/ns2/velodyne_points
/joint_states
--
which topic do you think is problematic?

aidadgy, 2024-10-12:
> Excuse me, I am running "roslaunch mins simulation.launch cam_enabled:=true lidar_enabled:=true". What is the cause of the error? How to fix it?
> 
> ![6676fd3ff17ca00e003eacc1acc1215](https://private-user-images.githubusercontent.com/103757590/283968332-38e66b24-d7c9-4d7f-9428-5818be4efbab.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg3MTU2MDIsIm5iZiI6MTcyODcxNTMwMiwicGF0aCI6Ii8xMDM3NTc1OTAvMjgzOTY4MzMyLTM4ZTY2YjI0LWQ3YzktNGQ3Zi05NDI4LTU4MThiZTRlZmJhYi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAxMlQwNjQxNDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00NzY2NzY3NWIwNTg3M2M3MWNjNjI1MmM4NjJlZjU3NmYwYTk5MWI1Y2M0NGM4M2Q2MTQ4MjUzOTAyNDFmNTAwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.gnJPtP-Vh_h4B-2tiLkdIanNIJH2eEzzkMyveWp4-Ow)
> 
> And I'm using "roslaunch mins serial.launch config:=kaist/kaist_LC path_gt:=urban30.txt path_bag:=urban30.bag "When running rosbag with your own kaist dataset, can the bag contain only the following topics? What are the topics of your bag?
> 
> ![a2d162de8c8437e781ab1e8ca21869d](https://private-user-images.githubusercontent.com/103757590/283968647-f6a2dff1-af3d-40a5-b5ab-6602d9eea75c.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg3MTU2MDIsIm5iZiI6MTcyODcxNTMwMiwicGF0aCI6Ii8xMDM3NTc1OTAvMjgzOTY4NjQ3LWY2YTJkZmYxLWFmM2QtNDBhNS1iNWFiLTY2MDJkOWVlYTc1Yy5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAxMlQwNjQxNDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02NzMwZDEyZGNkYjE0YzA0M2VjNzJhY2RjNWYwYTYzNTI4NzI5ZGFmZjQ5MWUxODc4ZjMzNDQxYjEyMDJmZjRkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.amlB5Zd60N3UiYU0V7PNwuhXL7gsrqfS2aXb3TN9XvU)
> 
> Thank you very much.

Hello author, I also encountered a similar problem as you. When I ran roslaunch mins simulation.launch cam_enabled:=true lidar_enabled:=true, the following problems occurred:
========Final Status========
Total procesing time: 21s
Total traveling time: 48s

RMSE average: 0.257, 0.117 (deg,m)
NEES average: 7.133, 2.380 (deg,m)
simulation: /usr/include/boost/thread/pthread/recursive_mutex.hpp:108: void boost::recursive_mutex::lock(): Assertion `!posix::pthread_mutex_lock(&m)' failed.
================================================================================REQUIRED process [mins_simulation0-2] has died!
process has died [pid 1327580, exit code -6, cmd /home/ubantu/catkin_ws/devel/lib/mins/simulation __name:=mins_simulation0 __log:=/home/ubantu/.ros/log/d0e201e6-885e-11ef-8f8b-41555d88daad/mins_simulation0-2.log].
log file: /home/ubantu/.ros/log/d0e201e6-885e-11ef-8f8b-41555d88daad/mins_simulation0-2*.log
Initiating shutdown!
================================================================================
[mins_simulation0-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
I wonder if the author can give some suggestions，thanks