rpng/MINS #14: how to implement with own hardware platform ?

mahammadirfan, 2023-12-08:
Hi, 
   Thanks for great work. I am wondering if i can use this MINS algorithm with the hardware like ouster, stereo camera etc ? is there any detailed documentation to follow ? 
Thanks in advance!

WoosikLee2510, 2023-12-13:
Hi, @mahammadirfan. Thanks for having an interest in MINS!
I will try to document some exemplary cases and tutorials on how to use MINS.
Meantime, can you tell me more about your system, please?

mahammadirfan, 2023-12-15:
@WoosikLee2510 Thanks for your reply, I am working with ouster lidar os1 with IMU, stereo camera, gps and want to fuse all the sensor like you have done in mins ...my use case is with a drone. Thanks!!

mahammadirfan, 2024-01-10:
@WoosikLee2510 Hey, I am trying to use the real world dataset example (urban30), but the ground truth i got one for this is in csv format, so i converted it to txt, Now i am getting some error ?? do you have a ground truth txt file for urban 30 dataset ? could you please share ? how to convert the csv to txt file ?
Thanks in advance!

mahammadirfan, 2024-01-11:
@WoosikLee2510 Hi
I am trying to use the kaist dataset utban 30 ....but getting an error when running the mins_slam...
1) I have downloaded the kaist dataset and followed the steps from kaist2bag 
https://github.com/rpng/kaist2bag
3) converted and got the bag file of 31.8 gb after merging all the individual bags.
4) when i am running the example: **roslaunch mins serial.launch config:=kaist/kaist_LC path_gt:=urban30.txt path_bag:=urban30.bag**
it gives error .....**[IW-Init] smooth fail -nan > 0.10703491   Duration: 0.301214 / 1267.945918**        

Looking forward to hear from you. Thanks
[image]

WoosikLee2510, 2024-01-26:
> @WoosikLee2510 Thanks for your reply, I am working with ouster lidar os1 with IMU, stereo camera, gps and want to fuse all the sensor like you have done in mins ...my use case is with a drone. Thanks!!

Hi- Sorry for my late reply, I was away from my desk for a while and now I am back. For Urban30 gt file, try using this.
[urban30.txt](https://github.com/rpng/MINS/files/14060634/urban30.txt)

WoosikLee2510, 2024-01-26:
> ![Screenshot from 2024-01-10 14-07-21](https://private-user-images.githubusercontent.com/51612103/295592767-3daff72a-5548-4429-8441-2481fcf2d2c4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDYyNDcwNDQsIm5iZiI6MTcwNjI0Njc0NCwicGF0aCI6Ii81MTYxMjEwMy8yOTU1OTI3NjctM2RhZmY3MmEtNTU0OC00NDI5LTg0NDEtMjQ4MWZjZjJkMmM0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTI2VDA1MjU0NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTlhZGQ1OTQzOGJkOTliYmEwYTM4MDU2NjU2OGU3MjliNTZhMzJhM2Y5YjEwNzBlYzZiNjAzZjI3YjAxMDkxMzUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.uFDgezlf24OlCtL8NZqcVk_VDOQJGYe812HUvSimKEo)

The error says the system tried to initialize from ground truth but it did not exist. You can either use the gt file I provided or disable initialization from gt [option](https://github.com/rpng/MINS/blob/master/mins/config/kaist/kaist_LC/config_init.yaml#L9)

WoosikLee2510, 2024-01-26:
> @WoosikLee2510 Hi I am trying to use the kaist dataset utban 30 ....but getting an error when running the mins_slam...
> 
> 1. I have downloaded the kaist dataset and followed the steps from kaist2bag
>    https://github.com/rpng/kaist2bag
> 2. converted and got the bag file of 31.8 gb after merging all the individual bags.
> 3. when i am running the example: **roslaunch mins serial.launch config:=kaist/kaist_LC path_gt:=urban30.txt path_bag:=urban30.bag**
>    it gives error .....**[IW-Init] smooth fail -nan > 0.10703491   Duration: 0.301214 / 1267.945918**
> 
> Looking forward to hear from you. Thanks ![Screenshot from 2024-01-11 17-55-01](https://private-user-images.githubusercontent.com/51612103/296008379-437ec90b-61d7-4f64-a768-a738befcbf62.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDYyNDcwNDQsIm5iZiI6MTcwNjI0Njc0NCwicGF0aCI6Ii81MTYxMjEwMy8yOTYwMDgzNzktNDM3ZWM5MGItNjFkNy00ZjY0LWE3NjgtYTczOGJlZmNiZjYyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTI2VDA1MjU0NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZlNDQzNjY3MzBlODM3YzE5YWVhNmIyMmM3ZTNhNzA1YWQ4NGFkM2M1YzZjNmEwODMyOTAyZjllOGYwMjg0MWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.uYIMWoRV1_CALZVOM7yTrNn20O-AauIaRNj_JqUEswo)

Hum... This is strange. Basically, the error is saying that some of the IMU reading were invalid (NaN). I can tell it from ```-nan > some number```. Could you check the IMU readings and see if any of them contains NaN?

wolf943134497, 2024-01-26:
WHEEL] Avg res/std: -5e-04/2e-03 | R std: 2e-04, Chi 100 %, mean: 1e-01, std 2e-01, # meas: 12702
[GPS0] Avg res/std:  1e-01/2e+00 | R std: 2e+01, Chi 100 %, mean: 9e-01, std 5e-01, # meas: 1863
1544579305.352 (125.017s) | q_GtoI = -0.004,-0.004,0.879,0.476 | p_IinG = -1911.683,2935.250,-5.906 | dist = 3421.74 (m)
Avg processing time(Curr Intr 3:15 Hz | Avg Intr 3:17 Hz): GPS: 0ms IMU: 0ms WHL: 0ms Total: 0ms / 56ms (125.1X)
Hz average: IMU 100.0 GPS0 5.0 WHL 100.0

[WHEEL] Avg res/std: -5e-04/2e-03 | R std: 2e-04, Chi 100 %, mean: 1e-01, std 2e-01, # meas: 12708
[GPS0] Avg res/std:  1e-01/2e+00 | R std: 2e+01, Chi 100 %, mean: 9e-01, std 5e-01, # meas: 1866
1544579305.419 (125.084s) | q_GtoI = -0.004,-0.005,0.879,0.476 | p_IinG = -1912.660,2936.750,-5.951 | dist = 3423.41 (m)
Avg processing time(Curr Intr 3:15 Hz | Avg Intr 3:17 Hz): GPS: 0ms IMU: 0ms WHL: 0ms Total: 0ms / 56ms (125.0X)
Hz average: IMU 100.0 GPS0 5.0 WHL 100.0

[WHEEL] Avg res/std: -5e-04/2e-03 | R std: 2e-04, Chi 100 %, mean: 1e-01, std 2e-01, # meas: 12714
[GPS0] Avg res/std:  1e-01/2e+00 | R std: 2e+01, Chi 100 %, mean: 9e-01, std 5e-01, # meas: 1866
1544579305.486 (125.151s) | q_GtoI = -0.005,-0.005,0.879,0.476 | p_IinG = -1913.558,2938.125,-5.947 | dist = 3425.07 (m)
Avg processing time(Curr Intr 3:15 Hz | Avg Intr 3:17 Hz): GPS: 0ms IMU: 0ms WHL: 0ms Total: 0ms / 56ms (125.0X)
Hz average: IMU 100.0 GPS0 5.0 WHL 100.0

[WHEEL] Avg res/std: -5e-04/2e-03 | R std: 2e-04, Chi 100 %, mean: 1e-01, std 2e-01, # meas: 12720
[GPS0] Avg res/std:  1e-01/2e+00 | R std: 2e+01, Chi 100 %, mean: 9e-01, std 5e-01, # meas: 1866
1544579305.552 (125.217s) | q_GtoI = -0.005,-0.005,0.879,0.476 | p_IinG = -1914.448,2939.497,-5.940 | dist = 3426.73 (m)
Avg processing time(Curr Intr 3:15 Hz | Avg Intr 3:17 Hz): GPS: 0ms IMU: 0ms WHL: 0ms Total: 0ms / 56ms (125.0X)
Hz average: IMU 100.0 GPS0 5.0 WHL 100.0

========Final Status========
Total procesing time: 0s
Total traveling time: 125s

================================================================================REQUIRED process [mins_bag-2] has died!
process has finished cleanly
log file: /home/gw/.ros/log/530bedfa-bc2a-11ee-b37c-4d6550c907d9/mins_bag-2*.log
Initiating shutdown!
================================================================================
[live_align_trajectory-3] killing on exit
[mins_bag-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Hi @WoosikLee2510   I use the newest code, I also encountered similar problem.  I only use imu wheel data.

zhh2005757, 2024-01-26:
I also want to use MINS in my own platform. The fusion of camera, imu and wheel odometer worked but lidar fusion has some problem. My lidar is velodyne 16 and I use the parameters of kaist dataset in MINS, but the trajectory has large drift even cannot maintain normal shape. Could your provide some detail about lidar parameters? Thanks!

mahammadirfan, 2024-01-26:
@WoosikLee2510 Thanks for your reply, Could you please explain or provide some details docs/steps for the calibration things like....which calibration tools should we use...i am using kalibr for stereo-imu....for cam-imu-lidar which tool should i use ?

Thanks and looking forward to hearing from you.

tejasps28, 2024-01-28:
> I also want to use MINS in my own platform. The fusion of camera, imu and wheel odometer worked but lidar fusion has some problem. My lidar is velodyne 16 and I use the parameters of kaist dataset [...]

@zhh2005757 
Hi, i am new to VINS, and I want to perform camera imu and wheel fusion. Can you please share how did you do the same ? Was it using MINS? 
Any information would be really helpful!! 
Thanks in advance

lnexenl, 2024-01-30:
> @WoosikLee2510 Thanks for your reply, Could you please explain or provide some details docs/steps for the calibration things like....which calibration tools should we use...i am using kalibr for [...]

I have runned MINS on some private datasets (Mono cam+Lidar+IMU+INS), but we calibrate sensors with our private tool. But there are some open-sourced tools you may use. However, there are no good open-sourced cam-imu-lidar calibration tools for use.

1. Lidar-Camera calibration [direct_visual_lidar_calibration](https://github.com/koide3/direct_visual_lidar_calibration)
2. Lidar-IMU Calibration [lidar_IMU_calib](https://github.com/APRIL-ZJU/lidar_IMU_calib)

These tools use target-less algorithms, thus you don't need a calibration board.

zhh2005757, 2024-01-30:
> @zhh2005757 Hi, i am new to VINS, and I want to perform camera imu and wheel fusion. Can you please share how did you do the same ? Was it using MINS? Any information would be really helpful!! [...]

Yes, I performed camera, imu and wheel odometer well by MINS in my private dataset. Of course good calibration (camera intrinsics and camera imu wheel extrinsics) is the first before testing MINS. In addition that, since the wheel odometer measurements of my dataset are linear velocities and angular velocities, I changed the type as "Wheel3DCen" in config_wheel.yaml and modified the topic subscription interface to ensure MINS to receive the wheel odometer measurements. Then MINS can work well in my dataset.

Also, I have performed camera-imu-wheel fusion in other frameworks like VINS-Mono. There are many corresponding open-source code that you can easily find.

wolf943134497, 2024-02-01:
@zhh2005757 hi， could you share your modified code or project of MINS?  
 Have you successfully tested other sensor combinations using dataset similar to kaist? like imu + wheel， imu + wheel + gps， imu + wheel + lidar or Camera-GPS-LiDAR-wheel-IMU .
 Thanks!

mahammadirfan, 2024-02-01:
@zhh2005757 Hey thanks for the reply....would be able to share your modified files for reference.
Thanks!

mahammadirfan, 2024-02-06:
@WoosikLee2510 Hey, I just checked and run the with the simulation example and real world dataset of  euros_mav ...v1-03-difficult.....where I dont see any loopclosure for the SLAM ? Whats the for loop closure not happening ? or this SLAM system doesn't support or do the loop closure ?

Thanks and looking forward to hearing from you!

WoosikLee2510, 2024-02-26:
> @WoosikLee2510 Hey, I just checked and run the with the simulation example and real world dataset of euros_mav ...v1-03-difficult.....where I dont see any loopclosure for the SLAM ? Whats the for [...]

Hi, @mahammadirfan. The current MINS does not support global visual loop closure (i.e., large correction of trajectory based on visual landmarks by revisiting the same place). However, you can input the loopclosure information (e.g., running another loopclosure detection thread) and use it to update MINS.

closed 2024-05-03 by WoosikLee2510

mahammadirfan, 2024-05-03:
@WoosikLee2510 
Hi,
Thanks for your reply again. I have a quick question....
when a run mins we get these topics 
**/mins/gps0/path
/mins/gps0/pose
/mins/imu/odom
/mins/imu/path
/mins/imu/pose**

So imu /path would be the final fusion path   and if i fuse cam + lidar  + imu +gps ?
and gps0/path is the only gps position path ?
Am i correct ? or gps0/path is the final fused path including gps ?

Thanks in advance!

WoosikLee2510, 2024-05-03:
The gps pose is the latest gps measurement received, and pose is the collection of poses.

Imu pose is the latest imu pose estimated (with update of other sensors), and the path is the collection of the imu pose.

aidadgy, 2024-11-20:
> @WoosikLee2510 Hi, Thanks for your reply again. I have a quick question.... when a run mins we get these topics **/mins/gps0/path /mins/gps0/pose /mins/imu/odom /mins/imu/path /mins/imu/pose**
>
> So [...]

Hello, may I ask how did you unify their coordinate system when comparing the running result and the truth value file? I noticed that the coordinate system in the urban.txt truth value file is different from the coordinate system in the result file

aidadgy, 2024-11-29:
> Hey, I just checked and run the with the simulation example and real world dataset of euros_mav ...v1-03-difficult.....where I dont see any loopclosure for the SLAM ? Whats the for loop closure not [...]
@mahammadirfan Hello, I would like to know how you compare and evaluate the estimated results of the program with the truth file urban30.txt, thank you