rpng/MINS #22: rviz configuration settings for simulation.launch, rosbag.launch and subscribe.launch

chowkamlee81, 2024-02-27:
@WoosikLee2510 @goldbattle @ghuangud @yangyulin @huaizheng @saimouli 

After launching the files simulation.launch, rosbag.launch and subscribe.launch, iam unable to see proper visualization in rviz for setting tf transformation coordinates and it is too confusing and time consuming. Can you plaese provide rvziz configuration for any of the one above launches so that i can roslaunch along with rviz appropriate settings. Requesting you kindly to do the needful

WoosikLee2510, 2024-02-27:
Please check https://github.com/rpng/MINS/blob/master/mins%2Flaunch%2Fdisplay.rviz if this is helpful

chowkamlee81, 2024-03-05:
[image]
Able to get the resut with the rviz pacage. Thanks a loy @WoosikLee2510 . Closing the issue