rpng/MINS #30: How to integrate loop closure?

mzahana, 2024-05-27:
Hi @WoosikLee2510 
Is it possible to integrate loop closure with MINS. My goal is to use MONO VIO + loop closure.

Thanks

WoosikLee2510, 2024-05-28:
Hi, @mzahana. 
You can use [UpdaterVicon](https://github.com/rpng/MINS/blob/master/mins/src/update/vicon/UpdaterVicon.h) to fuse loop closure information to filter. 
Say, your loop closure method in another thread detected loop closure and computed a constraint that can update the front end. You can convert the loop closure constraint to global pose information (like Vicon) with uncertainty (you may want to set a conservative value).

mzahana, 2024-05-28:
Thanks @WoosikLee2510 
Can the [ov_secondary](https://github.com/rpng/ov_secondary) be used to do the loop closure and feedback to MINS?

WoosikLee2510, 2024-05-28:
Note [ov_secondary](https://github.com/rpng/ov_secondary) does not provide information to update the front end (e.g., [OpenVINS](https://github.com/rpng/open_vins)). I think it can work if you create your own publisher to publish the loop closure information and use MINS to subscribe it as global pose information.

mzahana, 2024-05-28:
I see there is a topic published for the output odometry from the loop fusion node
https://github.com/rpng/ov_secondary/blob/c6b9e62a8bae453a230d024b4626906f62b5fb98/loop_fusion/launch/posegraph.launch#L27

Can this be fed to MINS?

WoosikLee2510, 2024-05-28:
Sure. you can add a subscriber that subscribes ([under here](https://github.com/rpng/MINS/blob/master/mins/src/core/ROSSubscriber.cpp)) to the topic with type, convert the message to [vicon type](https://github.com/rpng/MINS/blob/master/mins/src/update/vicon/ViconTypes.h), and pass it. to system.
I would be careful not to make the information overconfident because the loop closure information may be highly correlated to MINS and thus can easily hurt consistency.

mzahana, 2024-05-28:
I thought MINS already subscribes to vicon ? no? @WoosikLee2510

WoosikLee2510, 2024-05-31:
No, it is not implemented because full global pose information (ViCON) is not something the estimator usually has access to. 
In the case you do not want to implement the subscriber by yourself but have rosbag that has the pose information converted from loop closure, you can use/run [ros_bag](https://github.com/rpng/MINS/blob/master/mins/src/run_bag.cpp#L182-L193) which directly reads the messages from the bag and passes to estimator.

closed 2024-10-15 by WoosikLee2510