rpng/MINS #34: use mins to locate on a pre-built point cloud map

piedaochuan1234, 2024-07-12:
Thanks for your great work.

I would like to use mins for localization based on an pre-built point cloud map. In the original method of mins,  the map frame anchored to the latest IMU frame which can cause the map frame to be affected by the state estimation . Therefore, my idea is not to transform the point cloud map to the clone pose in real time, and frame will be like：
 map  --  C  --  C  -- I

 As a beginner, I am wondering if there are any suggestions or if the above idea is feasible?

lnexenl, 2024-07-12:
How do you want to utilize pre-built point cloud map? By registering Lidar observations to the cloud to the pre-built map or something else? I believe that you can consider the estimated poses from the pre-built map as outputs from a fake vicon sensor, and you don't need to do much modification to the code.

piedaochuan1234, 2024-07-13:
Yes, as you say, I want to register Lidar observations to the cloud with the pre-built map. I will learn how to implement it through Vicon.

Thanks

lnexenl, 2024-07-15:
I have done some work on prior lidar map assisted odometry, you may refer to [PPCA-VINS](https://github.com/lnexenl/PPCA-VINS). Hope this helps.

piedaochuan1234, 2024-07-16:
@lnexenl, Thanks for the quick reply and suggestions. I've tried using nano_gicp and ndt, but the CPU usage in the virtual machine is very high. I will try to estimate the global pose with a lightweight method, such as AMCL, as a fake Vicon measurements

closed 2024-07-17 by piedaochuan1234

ZYR2001, 2026-05-28:
> Yes, as you say, I want to register Lidar observations to the cloud with the pre-built map. I will learn how to implement it through Vicon.
> 
> Thanks

Hi, sorry to disturb you. I'm interested in doing something similar. Have you figured out this issue? Could we possibly discuss it?