rpng/MINS #36: use `run_bag` but very slow

WZG3661, 2024-07-26:
When I ran the `rosbag.launch`, the program runs very slow, it seems to run in real time.
Did I set some parameters wrong?
I just use camera and imu, set msckf points to 40, and slam points to 50

aidadgy, 2024-10-12:
Hello author, I also encountered a similar problem as you. When I ran roslaunch mins simulation.launch cam_enabled:=true lidar_enabled:=true, the following problems occurred:
========Final Status========
Total procesing time: 21s
Total traveling time: 48s

RMSE average: 0.257, 0.117 (deg,m)
NEES average: 7.133, 2.380 (deg,m)
simulation: /usr/include/boost/thread/pthread/recursive_mutex.hpp:108: void boost::recursive_mutex::lock(): Assertion `!posix::pthread_mutex_lock(&m)' failed.
================================================================================REQUIRED process [mins_simulation0-2] has died!
process has died [pid 1327580, exit code -6, cmd /home/ubantu/catkin_ws/devel/lib/mins/simulation __name:=mins_simulation0 __log:=/home/ubantu/.ros/log/d0e201e6-885e-11ef-8f8b-41555d88daad/mins_simulation0-2.log].
log file: /home/ubantu/.ros/log/d0e201e6-885e-11ef-8f8b-41555d88daad/mins_simulation0-2*.log
Initiating shutdown!
[mins_simulation0-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
I wonder if the author can give some suggestions，thanks

WoosikLee2510, 2024-10-15:
> When I ran the `rosbag.launch`, the program runs very slow, it seems to run in real time. Did I set some parameters wrong? I just use camera and imu, set msckf points to 40, and slam points to 50

Hi, @WZG3661 . I do not think your configuration (50 slam, 40 msckf) can slow down the process unless it is on an embedded device. Could you provide more information about your configuration and platform?

WoosikLee2510, 2024-10-15:
> Hello author, I also encountered a similar problem as you. When I ran roslaunch mins simulation.launch cam_enabled:=true lidar_enabled:=true, the following problems occurred: ========Final [...]

https://github.com/rpng/MINS/issues/31#issuecomment-2412902648

closed 2024-11-06 by WoosikLee2510