rpng/MINS #39: Adopt Livox lidar

qiqzhang, 2024-08-11:
Thanks for your wonderful job. When I adopt livox lidar with point time relative to first point time, header time use first lidar point time, but traj is drift than undistorted pointcloud. what should i do to adopt livox lidar

qiqzhang, 2024-08-11:
I change some code  to adopt livox， like this, 
success = state->get_interpolated_pose(lidar_inL->time + lidar_inL->pointcloud->points[i].curvature * 1e-6, RGtoIi, pIiinG);
 
if ((*it)->time + state->lidar_dt.at((*it)->id)->value()(0) + (*it)->pointcloud->points[(*it)->pointcloud->points.size() -1 ].curvature * 1e-6 > state->newest_clone_time()) {
       it++;
      continue;
}

what should i do next

qiqzhang, 2024-08-12:
[image]
[image]

muttistefano, 2024-09-26:
Hi @qiqzhang ,
Are you using the livox mid360 or another model?
I'm asking because the mid360 gives me some problems.
Thanks

qiqzhang, 2024-10-02:
> Hi @qiqzhang , Are you using the livox mid360 or another model? I'm asking because the mid360 gives me some problems. Thanks
I use models that like mid360, original undistorted pointclouds maybe adopted to mid360,  original way supposes every point has 
fixed time gap,  I tried to adopt mid360, but not work well. @WoosikLee2510 Could you give us advices to get correctly results.

WoosikLee2510, 2024-10-15:
Hi, @qiqzhang. Thanks for bringing up this issue. I will check to see what the problem is. Definitely, not working well with simulation is a bug issue.

jeannotes, 2025-04-07:
hi, any results