rpng/MINS #40: Map Erases itself randomly after a certain amount of Time

BearrGod, 2024-09-26:
Hello. I'm using Mins to Construct a dense map of an indoor environment.  So i've set the threshold of the map_decay_dist to 1000 and map_decay_time to 3600 to avoid map erasure overall. But after some time the map gets erased. Does someone have an explanation for that ? Or perhaps are there other options i'm missing here.

WoosikLee2510, 2024-10-15:
you also need to disable `map_do_downsample: false` as it sparsifies the lidar map.

BearrGod, 2024-10-17:
I figured the reason for the map erasure lied in the fact that i lack computing power, and from [here](https://github.com/rpng/MINS/blob/ed6cf522845e09c8ee33da58f27aad46aeba6337/mins/src/update/lidar/UpdaterLidar.cpp#L90) since the process cannot be done in real time , the map is reinitialized. In my case i just made it +10 seconds to delay the erasure ,  but would be nice to have it as a parameter.

closed 2024-10-23 by BearrGod