rpng/MINS #41: Livox Mid-360

muttistefano, 2024-09-26:
Hi all, and thanks for the great work.
Is there a way to make the MID360 and its 6-axis IMU work with MINS?
I am experiencing a lot of drift; when I turn the lidar, the odometry moves as if I were accelerating in the direction I rotated the sensor.
Is this just because of the parameter tuning and calibration?
Is the lidar rectification method suitable for MID360?
Thanks

WoosikLee2510, 2024-10-15:
Could you please provide your config and rosbag file so I can look?

WoosikLee2510, 2024-11-06:
> > Could you please provide your config and rosbag file so I can look?
> 
> Thanks for your work, could you please tell me the imu used in this work is 6 axis or 9-axis

https://github.com/rpng/MINS/issues/45#issuecomment-2443323115

closed 2024-11-06 by WoosikLee2510

jeannotes, 2025-04-07:
hi, for example thiese bags:
https://connecthkuhk-my.sharepoint.com/personal/zhengcr_connect_hku_hk/_layouts/15/onedrive.aspx?id=%2Fpersonal%2Fzhengcr%5Fconnect%5Fhku%5Fhk%2FDocuments%2Ffast%2Dlivo2%2Ddataset&ga=1