rpng/MINS #45: IMU and wheel odometer of this work

supersupercoconut, 2024-10-25:
Thanks for your valuable work! I have some questions regarding this SLAM algorithm:
1. Could you please clarify whether this work utilizes a 6-axis or 9-axis IMU sensor? 
2. How the wheel odometry impacts the Z-axis accuracy of the system? Does it enhance or diminish the vertical accuracy of the pose estimation?

WoosikLee2510, 2024-10-29:
Hi, @supersupercoconut .
1. When you say 6 axis is it mean {gyro, accel}, and 9 axis means {gyro, accel, mag}? If that's the case, MINS uses a 6-axis. If you want to include {mag}, I suggest using {mag} separately as an orientation update method. 
2. MINS supports multiple assumptions for wheel odometry (see: https://github.com/rpng/MINS/blob/ed6cf522845e09c8ee33da58f27aad46aeba6337/mins/src/update/wheel/WheelTypes.h#L27-L32). One common mode is `Wheel3DAng` which allows you to leverage planar motion constraint which helps reducing z-axis drift.

supersupercoconut, 2024-10-29:
> Hi, @supersupercoconut .
>
> 1. When you say 6 axis is it mean {gyro, accel}, and 9 axis means {gyro, accel, mag}? If that's the case, MINS uses a 6-axis. If you want to include {mag}, I suggest using [...]

Yes, when I refer to the 6-axis IMU, I'm talking about the gyroscope and accelerometer. I would like to confirm if the data from the /joint_state topic is derived from the fusion of the chassis IMU and a wheel odometer?

WoosikLee2510, 2024-11-06:
In my experience, there are different cases. For example, the KAIST dataset provides direct wheel angular velocity, which you can multiply by wheel radius to get each wheel's velocity. On the other hand, Husky was providing odometry fused with IMU, as you said.

closed 2024-12-04 by supersupercoconut