rpng/MINS #46: terminate called after throwing an instance of 'class_loader::LibraryUnloadException

aidadgy, 2024-11-02:
Hello author, I reconfigured the computer environment and finally ran a new error:terminate called after throwing an instance of 'class_loader::LibraryUnloadException'   what():  Attempt to unload library that class_loader is unaware of.May I ask what caused this error

WoosikLee2510, 2024-11-06:
Re: email reply
The error is something I also get all the time. 
The error does not stop the system while running; it comes from terminating classes at the end of a run (I suspect related to some visualization).
Since it will not affect performance, I recommend ignoring it—or let me know if you can resolve the issue.

closed 2024-11-06 by WoosikLee2510

aidadgy, 2024-11-14:
> Re: email reply The error is something I also get all the time. The error does not stop the system while running; it comes from terminating classes at the end of a run (I suspect related to some [...]

Thank you for your answer