rpng/MINS #48: problem about 他和

aidadgy, 2024-11-22:
Hello author, I conducted experiments using the publicly available dataset urban30.txt and recorded the results of the true and estimated values, which are path_data.txt path_data2.txt. I evaluated them using evo_ape tum path_data.txt path_data2.txt - a-p - s. The results are as follows:
file:///home/alading/%E5%9B%BE%E7%89%87/mins/2024-11-21%2017-12-52%20%E7%9A%84%E5%B1%8F%E5%B9%95%E6%88%AA%E5%9B%BE.png
But I found that the RMSE index is 9.25, which cannot achieve the accuracy mentioned in your paper，Do you know what the reason is，thanks
file:///home/alading/%E5%9B%BE%E7%89%87/mins/2024-11-10%2014-10-26%E5%B1%8F%E5%B9%95%E6%88%AA%E5%9B%BE.png

aidadgy, 2024-11-30:
> Hi, I think you have misunderstood the meaning. The author said in the paper the ATE is per 1km, not entire trajectory. Hope this can help you.
Yes, I have also noticed that, and I would like to ask how you record the estimated result and the truth result. I used to record the corresponding topic data before, but I don't know whether it is appropriate, because the data in the truth topic seems to be constantly updated and changed, thank you

Happy-ZZX, 2024-12-08:
> Hi, I think you have misunderstood the meaning. The author said in the paper the ATE is per 1km, not entire trajectory. Hope this can help you.
> Yes, I have also noticed that, and I would like to ask [...]

Yes, I also notice that the Ground Truth is keep changing in Rviz. But if you draw the estimated result and ground truth, it is right. To draw the trajectory result, there are plenty of toolboxes you can use, like Openvins tools, EVO evaluate toolboxes.

aidadgy, 2024-12-08:
> Yes, I also notice that the Ground Truth is keep changing in Rviz. But if you draw the estimated result and ground truth, it is right. To draw the trajectory result, there are plenty of toolboxes you can use, like Openvins tools, EVO evaluate toolboxes.

I'm sorry, I can't understand what you said: "But if you draw the estimated result and ground truth, it is right." What does it mean? Do you mean that you do not need to care about this phenomenon, and still record their corresponding topic data for comparison? I am curious why the truth value is changing, is it some provisions made in the source code, thank you

Happy-ZZX, 2024-12-08:
> I'm sorry, I can't understand what you said: "But if you draw the estimated result and ground truth, it is right." What does it mean? Do you mean that you do not need to care about this phenomenon, [...]

Sorry for the confusion. Yes, you can ignore this issue as it’s just a display bug. I suspect the root cause is a third-party library in Ov_eva. Recently, I’ve been working on resolving this problem. However, since I’ve been busy with other tasks, I’m not sure when I’ll be able to identify and fix the bug.

aidadgy, 2024-12-11:
> Sorry for the confusion. Yes, you can ignore this issue as it’s just a display bug. I suspect the root cause is a third-party library in Ov_eva. Recently, I’ve been working on resolving this [...]

Thank you for your answer. As you record the content of the truth topic, you will find that the previous position results are constantly updated. Have you noticed this phenomenon? The result of truth value is constantly updated and changed, so can we still use the truth value recorded by this topic to evaluate, or are you directly using the truth value file provided by the author

Happy-ZZX, 2025-01-27:
> Thank you for your answer. As you record the content of the truth topic, you will find that the previous position results are constantly updated. Have you noticed this phenomenon? The result of [...]

Hi,  Sorry for the late reply. I get your meaning, finally. If you try to use EVO to draw the trajectory figure, I suggest you first use a script that transforms the output file format into TUM or Eruoco format. Then use EVO to draw the trajectory.